sidecartridge/atarist-sidecart-raspberry-pico #28: After resetting to the configurator, wifi often doesn't work without reconfiguring

Cwiiis, 2023-10-14:
After resetting the sidecart, upon launch I find that often wifi won't connect and just displays error - it won't reconnect until I go into wifi configuration and re-enter details, even though the details already exist and are correct in the SidecarT configuration screen.

Perhaps there's some sensible heuristic to retry connecting if it fails sometimes, or failing that, maybe manual disconnect/connect shortcuts would be helpful.

diegoparrilla, 2023-10-14:
I've observed this error and suspect it's tied to potentially corrupted data in the Infineon CYW43439 WiFi chip, which may not be thoroughly cleared by the API upon device reset. I will delve into this issue to identify a possible solution, such as resetting the WiFi chip upon error detection, avoiding a full device restart.

In the meantime, a power cycle for your computer may serve as a workaround. Allow a few seconds before turning it back on as certain Atari ST models may continue to supply power to cartridge devices briefly due to capacitors.

diegoparrilla, 2023-10-14:
I'm testing with a manual retry when the connection fails.

closed 2023-10-22 by diegoparrilla