sidecartridge/atarist-sidecart-raspberry-pico #29: Feature request: Cubase dongle emulation

Cwiiis, 2023-10-14:
It would be great if the SidecarT could be configured to emulate a Cubase dongle. I would be happy to work on this, but would need some guidance on where to start - I'd have thought it ought to be relatively trivial, but maybe it isn't as simple as I think :)

diegoparrilla, 2023-10-14:
Definitely, this is one of the things that SidecarT could do. 

I have to finish the developer documentation here: https: //docs.sidecart.xyz and I will try to explain how to use the ROM emulator and the commands to implement custom firmware enhancements.

May be the people in the Atari Forum can help us. There is a very good thread about this topic.

Ka-non, 2023-10-17:
have you de ROM file of the dongles? 
I can test it

Cwiiis, 2023-10-17:
> have you de ROM file of the dongles? I can test it

The dongle isn't a plain ROM, it's a device that performs various operations on inputs that the program expects particular outputs for. I've attached the .jed export for reference.

[srd_ok.jed.txt](https://github.com/diegoparrilla/atarist-sidecart-raspberry-pico/files/12927707/srd_ok.jed.txt)

Ka-non, 2023-10-17:
i know... but there is info about extract de ROM file of the dongle, an then, upload to sidecart and try...
[https://www.atari-forum.com/viewtopic.php?t=20130&start=350](url)

Cwiiis, 2023-10-17:
> i know... but there is info about extract de ROM file of the dongle, and then, upload to sidecart and try... [https://www.atari-forum.com/viewtopic.php?t=20130&start=350](url) (i believe)

The Cubase dongle isn't a ROM. A ROM is just data, you read an address and it returns a fixed piece of data. The dongle reads inputs (sort of), performs operations and gives outputs. It will require programming a binary to run on the SidecarT to emulate its behaviour. It won't just be a file that you upload as a ROM.

Ka-non, 2023-10-17:
ook.

diegoparrilla, 2023-10-17:
I am not an expert in this, but probably the JED implements some kind of cryptographic function. If we can reverse engineer it I think we could create a firmware for it. I will review the JED file, maybe I can see something...

diegoparrilla, 2023-10-17:
Ok. I have read the file and need more context, but I think is doable in a reasonable timeframe. Where can I find a Cubase version that I can test?

Ka-non, 2023-10-17:
you need a cubase version without "hack"...BUT there are differents versions of dongles, which is yours?

Cwiiis, 2023-10-17:
> Ok. I have read the file and need more context, but I think is doable in a reasonable timeframe. Where can I find a Cubase version that I can test?

If you let me know how best to contact you I can help you out - I'm Cwiiis on all social media and forums. Afk ATM, but back later.

closed 2023-10-22 by diegoparrilla