sidecartridge/atarist-sidecart-raspberry-pico #38: Configurable Virtual Floppy Emulation Startup

diegoparrilla, 2023-10-24:
## Current Behavior
At present, the Floppy Emulator invariably initiates as a Virtual A unit, undergoing the subsequent stages:
- **[NEW in v0.0.11]** The emulator verifies if a floppy is connected. Absent that, it establishes a virtual A drive placeholder.
- The Physical drive A is reconfigured and recognized as Physical drive B.
- The system attempts booting from the virtual drive A.

Due to this setup, users who wish to operate the physical drive A have to access it as unit B. This limitation impedes users from booting using the physical drive and could render specific software inoperative.

## Proposed Enhancement
We propose granting users the flexibility to decide between the default operational sequence detailed above and an alternative where they can allocate the virtual drive to unit B, enabling them to boot from the physical A drive.

**Bonus:** An optional feature could be a flag, allowing users to opt whether or not to boot from the virtual floppy, even when designated as the A drive.

RetroTechN00b, 2023-10-27:
The optional feature not to boot from the virtual drive sounds very desirable. Combined with the read/write capability on ST-Images this allows to use the virtual (ST-Image) drive to easily store data from an application booted from the physical drive.

diegoparrilla, 2024-04-03:
New floppy emulator can choose to boot or not from the virtual disk, bypassing virus detection code in the boot sector.

closed 2024-04-03 by diegoparrilla