sidecartridge/atarist-sidecart-raspberry-pico #48: Create an Empty Floppy Disk in Read/Write Floppy Emulation

diegoparrilla, 2023-10-31:
## Description

Currently, users looking to copy content to a floppy disk image are required to work with pre-existing images in the `/floppies` folder. This approach is not only inconvenient but also prone to errors, especially when managing multiple images or ensuring data integrity.

To streamline this process and enhance user experience, we propose the introduction of a feature that allows the creation of new, empty floppy disk images directly within the system. This feature will be integrated into the existing workflow, offering an efficient and user-friendly way to generate new .ST image files without the need to rely on external tools or pre-existing images.

### User Workflow

Upon selecting the option to create a new empty floppy disk image, the system will guide the user through a simple, interactive process:

1. **Prompt for New Floppy Image Name**: The user will be asked to provide a unique name for the new floppy disk image. This name will be used to identify the image file within the `/floppies` directory.

2. **Select Number of Sides**: The user can specify the number of sides for the floppy disk. This choice determines the disk's storage capacity and is critical for ensuring compatibility with the intended use case.

3. **Choose Number of Sectors**: The user will choose the number of sectors per track. This decision impacts the total storage capacity of the floppy disk and should be made based on the specific requirements of the content to be copied.

### Post-Creation Actions

Once the user completes the above steps, the system will execute the following:

- **Create the New .ST Image**: The system will generate a new, empty .ST floppy disk image based on the provided specifications. This image will be automatically saved in the `/floppies` folder with the designated name.

- **Automatically Mount the New Image**: The newly created floppy disk image will be automatically mounted. This ensures that it is immediately available for use, streamlining the user's workflow.

- **Reboot in Read/Write Mode**: To facilitate immediate use, the SidecarT system will reboot into Read/Write mode with the new floppy disk image mounted and ready for content copying.

### Conclusion

This new feature will significantly improve the flexibility and efficiency of the floppy emulation process. By enabling users to create bespoke, empty floppy disk images on demand, it eliminates the dependency on pre-existing images and reduces the potential for errors. This enhancement will be particularly beneficial for users regularly dealing with a variety of content and requiring custom storage solutions.

diegoparrilla, 2023-10-31:
Hatari code is GPL, we are GPL. So...

https://github.com/hatari/hatari/blob/master/src/createBlankImage.c

RetroTechN00b, 2023-10-31:
Very interesting feature. I was about to ask for it, but you are simply too fast ;-)

When mounting the new empty ST image, it would be helpful to be able to mount it as drive B:. At the same time mounting a different ST image as drive A: would be helpful. 

The use case I have in mind for this scenario is my 520ST without physical floppy. I would like to mount a productivity program from the ST image emulated in drive A: (read only), while the newly created ST image emulated in drive B: (read write) can be used to store the data created. Makes sense?

diegoparrilla, 2023-11-02:
Yes, it make sense. I have created this enhancement that will have to be developed first (not particularly complex): https://github.com/diegoparrilla/atarist-sidecart-raspberry-pico/issues/51

closed 2023-11-03 by diegoparrilla