sidecartridge/atarist-sidecart-raspberry-pico #59: Remote File decompression

a1exh, 2023-11-15:
There are various compressed file formats that are practically impossible to use with the ST 8MHz 68k but the RP2040 could decode them and present to the ST in a more usable format (e.g. RAW)

e.g. MP3 Audio

How best to implement? Perhaps start by emulating a stereo sound digitiser cartridge that existed? 

https://temlib.org/AtariForumWiki/index.php/Sound_Samplers

Some file selector to choose the remote file to start streaming?

diegoparrilla, 2023-11-18:
Interesting, I will have a look at this. Sounds like something perfect for a micro at 150Mhz like the RP2040.

diegoparrilla, 2024-01-19:
I think a set of features could be available without writing a specific firmware for it. Starting in version v1.0.0, all the features with their own firmware will be consolidated in the "Enchilada" firmware; it should be part of the new firmware model.