sidecartridge/atarist-sidecart-raspberry-pico #60: (WiFi) Configuration by file

ojaksch, 2023-11-18:
Would it be possible to have a text file somewhere to predefine configuration parameters?
My "issue" is that my WiFi password is very long, strong and therefore very difficult to enter.

diegoparrilla, 2023-11-18:
Very reasonable request indeed. Added to the next firmware release

RetroTechN00b, 2023-11-21:
Excellent idea. My password is quite long as well. Is there a limit of 42 character for the password? I think that was my conclusion.
My I add the suggestion to add a configuration parameter to hide the WiFi Password, so it would show up as a row of asterisks in the configuration program? I know it is a small gain in security as long as it is stored in plain text in a file.

ojaksch, 2023-11-21:
Looks like that WPA2 limits the [password length to 63 characters](https://stackoverflow.com/questions/18006390/why-is-the-wpa2-psk-key-length-limited-to-63-characters). But despite of that would it be imho useful **not** to set an input limit in favor for new or "other" WiFi future credentials/technics.

diegoparrilla, 2023-11-21:
It's a restriction of the user interface, the password in the file will allow much larger passwords.

closed 2023-11-22 by diegoparrilla

ojaksch, 2023-11-22:
Thank you for the implementation! Will try to compile this for myself and to test and enjoy this asap.
👍️