sidecartridge/atarist-sidecart-raspberry-pico #95: can you upload production files for the pcb?

PPZ-00, 2024-03-08:
Hello, 
Could you include production files for the PCB? Great project but I am kind of stuck with the schematics. :)

RetroTechN00b, 2024-03-09:
I would not think this a good idea since it would open the field for a wave of low quality china-copies of the SidecarT. The beauty of this project is the quality and speed of the whole package of hardware and firmware. Low quality clones with the goal of making money out of other peoples work without contributing to the project does not help but potentially creates confusion and incompatibilities. 

This comment does not mean I would consider you @PPZ-00 to be one of these cloners. All I want to say is publishing the production files will simplify the unethical approach described above.

diegoparrilla, 2024-03-10:
The subject of sharing the SidecarT PCB design files has been a recurring topic in our Discord discussions. Regrettably, I must reiterate that I will not be making the PCB files available for several reasons.

The journey of creating the SidecarT was embarked upon primarily as a means for me to deepen my understanding of electronics. Coming from a background in software engineering, where I've developed a vast variety of solutions, venturing into hardware was a whole new terrain for me. My objective was to bridge my knowledge gap in hardware design by learning from both seasoned electronic engineers and my peers in software engineering. The process of seeing my design materialize on somebody else PCB is a significant part of this learning curve.

It's important to note that the SidecarT design is subject to licensing restrictions that explicitly prohibit commercial use. While I wholeheartedly encourage enthusiasts to build their own SidecarT for personal enjoyment or to share within their circles, I urge those seeking profit to respect the intentions and efforts that fueled this project. Any financial gains should duly acknowledge and support the original creator's contributions.

I am in favor of individuals, friends, communities, and open-source developers creating their own versions of the SidecarT. However, this project is not suited for those whose participation in the community is defined by commercial fabrication and sales. My long-standing involvement in open-source communities, both as a hobbyist and professionally, has equipped me with the knowledge and resources necessary to uphold the SidecarT's licensing terms.

Moreover, the development of the SidecarT has encompassed significant software effort, which is another reason to limit access to the product to ensure it remains within a circle of genuine enthusiasts and contributors, rather than mere spectators.

closed 2024-03-10 by diegoparrilla

PPZ-00, 2024-03-10:
> <diegoparrilla/atarist-sidecart-raspberry-pico/issue/95/issue_event/12064181236
> @github.com>

Hello,

I understand absolutely your attitude, if you would share freely all
production files, soon every store would sell it in order to make profit.

I actually bumped on a project some weeks ago when one fellow from Greece
made a video presentation of your cart.

I liked it so I wanted to order one from you directly, but unfortunately
you are not shipping to Greece anymore.
I don't understand why but you probably have a good reason.
So I decided to make cartridges for myself.
But then I saw there is only schematics of it.
I am an electronic engineer, but I thought to contact you first, maybe you
make my life easier and share the production files with me 😄

Anyway, if you are willing to share files only with me, I can assure you I
will not share them to anyone else.
I would eventually make small batch like 5-10 pieces for friends here in
Greece.
You would get your share of course, as it is your project.

Otherwise, please consider shipping to Greece again, I would take 3-5
cartridges :)

Thank you very much.

Best regards,
Peter

нед, 10. мар 2024. 02:07 Diego Parrilla ***@***.***> је
написао/ла: