sidecartridge/atarist-sidecart-raspberry-pico #103: No support for filenames with swedish characters on HD

Pny, 2024-04-25:
The harddrive emulation does not support filenames that includes Swedish/German characters, such as åäö and ÅÄÖ, (the a's and o's with dots and rings). If trying to copy such file onto the emulated HD an error message appears.

diegoparrilla, 2024-04-26:
GEMDRIVE only supports the "Legal Character in File Names" as described in the [ATARI GEMDOS REFERENCE MANUAL (1986), FILENAMES section (page 6)](http://www.bitsavers.org/pdf/atari/ST/Atari_ST_GEM_Programming_1986/GEM_0042.pdf
).

I strongly recommend against using unsupported characters, lowercase letters, and Long File Names (LFN) because they can crash your computer.

closed 2024-04-26 by diegoparrilla

RetroTechN00b, 2024-04-28:
Good to know, I was not aware Umlaute (äöüß) are not supported by Atari GEMdos. Typical US issue ;-) ... luckily I'm old enough to be trained on avoiding non-US characters, blanks, longer than 8.3 and so on in file names.

Pny, 2024-04-29:
I've never knew that either... Back in the day, I used those characters all the time. And the GEMDOS in my swedish TOS happily create, open, read, write and renames such files. My physical HD from that time has many such files. Even Hatari emulator suppport them, (but they doesnt show up as ÅÄÖ-characters on the PC). Only problem I've seen so far is SidecarT that corrupts the entire SD-card if I try to copy them to its HD-emulator... Had to reformat my SD-card a duzen of times while trying to move file structures from my physical SCSI-drive to my computer using the SidecarT.