sidecartridge/atarist-sidecart-raspberry-pico #121: Select button broke off board

Fab4583, 2024-07-15:
On the second day of use, the Select button broke off the board. The button design is not strong enough. I recommend a tactile button mounted flat on the board. I will repair my board by soldering a tactile button to the surface.

diegoparrilla, 2024-07-15:
Hi @Fab4583,

I'm sorry to hear that. Please send me an email to sidecart [AT] gooddatalabs.com (or any of the direct ways to contact https://sidecartridge.com/contact/) with your order number, and I will ship a new board.

closed 2024-07-15 by diegoparrilla

Fab4583, 2024-07-16:
I repaired the unit by installing a tactile button on the surface. A replacement board is not needed.

Perhaps a dot of epoxy on the back side of the switch will prevent this from happening.

diegoparrilla, 2024-07-16:
Sorry to insist. It's important for me that you have the best possible experience. Don't be shy and send me your order and I will send you a new one! :pray: 

We have been talking about this design issue in the Discord server: https://discord.com/channels/1160868666162823218/1160870626496626750/1262789203746623531

Feel free to joIn!