sidecartridge/atarist-sidecart-raspberry-pico #129: Second floppy disk drive (B:) emulation

danboid, 2024-08-17:
Do you think it will be possible for sidecart to emulate a second (or third/fourth) floppy drive?

danboid, 2024-08-19:
It seems it is not possible for the sidecart to emulate a cartridge and a hard disk simultaneously so I'd presume its not currently possible to emulate two floppy drives either.

Is/will it be possible to use one or both of the USB ports for a keyboard or mouse whilst emulating a disk or can it only emulate one device at a time?

danboid, 2024-08-19:
I have just changed the title of this issue after looking through the sidecart config menu options and noticing that it does seem to support emulating 2 floppy drives, simultaneously? I've not tested using 2 floppies yet.

So the issue now is that the "Emulate Floppy" menu should let you configure drive B in the same way it enables the user to pick an image via a list rather than having to manually enter the filename in the SidecarT config menu.

diegoparrilla, 2024-09-13:
Implemented in v0.0.18

closed 2024-09-13 by diegoparrilla