sidecartridge/atarist-sidecart-raspberry-pico #130: Software requests

danboid, 2024-08-17:
There are a few programs I'd like to see added to the sidecart floppy database please but I've not got anywhere to host them.

I hope I'm OK to make a few requests? I'm pretty sure these will be the only ones.

Deluxe Paint ST

https://www.atarimania.com/utility-atari-st-deluxe-paint-st_27258.html

At least the first disk.

ProTracker

https://www.atarimania.com/utility-atari-st-protracker-st_30690.html

Then people could run the same tracker on their modern computer as their ST a bit easier.

https://16-bits.org/pt2.php

You should try to appease us UNIX greybeards with some vim action. Of course I want to use vim natively on my ST!

https://github.com/simonlaszcz/vimtos

Thanks!

diegoparrilla, 2024-09-13:
You can submit the software as a PR: https://docs.sidecartridge.com/sidecartridge-multidevice/publicfloppydb/

danboid, 2024-09-15:
I would submit a PR if I had somewhere to host the images that uses http but all of those sites use https.

diegoparrilla, 2024-09-15:
Sorry, the RP2040 is not powerful enough to run https and the rest of the code of the firmware. Probably RP235x will be able to handle https.

danboid, 2024-09-15:
That's understandable but it would be cool if you had a FTP site we could upload disk images to for such situations.