sidecartridge/atarist-sidecart-raspberry-pico #133: WiFi password length should be 63? SSID should be 32?

shooking, 2024-08-24:
https://github.com/sidecartridge/atarist-sidecart-raspberry-pico/blob/fdab25c8789836367cc104ec968d853206927bea/romemul/include/network.h#L78

I couldn't work out why my 63 char WPA2 password doesn't work. I thought it was word wrapping during input. 
But it seems the max allowed password length is 36? Maybe 63 got transposed to 36? But then the SSID length also seems to be 36 rather than 32.

Did I miss something here?

diegoparrilla, 2024-09-12:
Definitely, it is a bug. I will fix it in the v0.0.18 release. 

Have you tried the `.wifipass` configuration file? It bypasses the WifiNetworkAuthInfo structure.

The 36 value is due to padding:
- 32 bytes for SSID
- 1 byte for NULL termination of the string
- 3 bytes for padding the content. RP2040 needs a 4-byte alignment in memory. Since I share this structure with Atari ST (which only needs 2 bytes of memory alignment) in memory, I have to make this rounding in the code very often.

shooking, 2024-09-12:
I have been looking into this more as a way to try to familiarise myself with your code.
Yes the .wifipass _sort_ of gets me out of jail. My current workaround is to use phone as a hotspot - it doesnt have a 63 char passphrase.

I am investigating if one can use snprintf to limit to the array size - ignore need for null terminator at all.
I see there are two network.h files to be wary off.
One in configuration/src/include/network.h for firmware.
One in romemul/include/network.h

I moved all the word fields 1st in the struct. Then surely who cares where the char arrays go?

If you want to allow for an additional \0 would it be clearer to use

```
char ssid[MAX_SSID_LENGTH + 1];
```

Similarly, if one refactors connection_data say to
```
typedef struct connection_data
{
    uint16_t wifi_auth_mode;                        // WiFi auth mode (WPA, WPA2, WPA3, etc)
    uint16_t wifi_scan_interval;                    // WiFi scan interval in seconds
    uint16_t network_status_poll_interval;          // Network status poll interval in seconds
    uint16_t network_status;                        // Network connection status
    uint16_t file_downloading_timeout;              // File downloading timeout in seconds
    uint16_t padding;                               // Padding to align to 4 bytes
    char ssid[MAX_SSID_LENGTH + 1];                     // SSID to connect
    char ipv4_address[IPV4_ADDRESS_LENGTH];         // IP address
    char ipv6_address[IPV6_ADDRESS_LENGTH];         // IPv6 address
    char mac_address[MAX_BSSID_LENGTH];             // MAC address
    char gw_ipv4_address[IPV4_ADDRESS_LENGTH];      // Gateway IP address
    char gw_ipv6_address[IPV6_ADDRESS_LENGTH];      // Gateway IPv6 address
    char netmask_ipv4_address[IPV4_ADDRESS_LENGTH]; // Netmask IP address
    char netmask_ipv6_address[IPV6_ADDRESS_LENGTH]; // Netmask IPv6 address
    char dns_ipv4_address[IPV4_ADDRESS_LENGTH];     // DNS IP address
    char dns_ipv6_address[IPV6_ADDRESS_LENGTH];     // DNS IPv6 address
    char wifi_country[4];                           // WiFi country iso-3166-1 alpha-2 code. 2 characters. Example: ES. 2 extra characters for padding
} ConnectionData;
```
doesnt that handle the alignment for you? I am research if the code make serialization assumptions or if it will just do the right thing.

Are the IPV4_ADDRESS_LENGTH also supposed to be null terminated - or can we not use sizeof(cd.ssd) to be MAX_SSID_LENGTH + 1?

shooking, 2024-09-12:
Ah the Pico needs the 32 bit alignment - got it!
Hmmh. So making the say MAX_WPA2_LENGTH = 64 gets the 63, a space for \0 AND makes it 4 byte aligned. Kinky.

shooking, 2024-09-12:
I think I can see you are working this bug. I managed to hack it - took me a while to understand why corruption was happening after 39th byte. Ah MAX_SSID_LENGTH + 2 !!

I can see you are adding some nice macro like

void network_swap_auth_data(uint16_t *dest_ptr_word)
{
    // +2 is for the auth_mode type
    CHANGE_ENDIANESS_BLOCK16(dest_ptr_word, MAX_SSID_LENGTH + MAX_PASSWORD_LENGTH + 2);
}

Dont you think you could use
```
void network_swap_auth_data(uint16_t *dest_ptr_word)
{
    // +2 is for the auth_mode type
    CHANGE_ENDIANESS_BLOCK16(dest_ptr_word, sizeof(WifiNetworkAuthInfo));
}
```
?? there is similar possible change for the 
```
{
    // +2 is for the auth_mode type
    //swap_words(dest_ptr_word, MAX_SSID_LENGTH + MAX_SSID_LENGTH + 2);
//    swap_words(dest_ptr_word, MAX_SSID_LENGTH + MAX_WPA2_LENGTH + 2); // I called it MAX_WPA2_LENGTH and made it 64
    swap_words(dest_ptr_word, sizeof(WifiNetworkAuthInfo));
}
```

I also had to change the print_table to use valueStr[MAX_STRING_VALUE_LENGTH].
Similarly in configuration().

The password max len in network_selector() and the read_string parameter.

And mirror the struct changes in the network.h of the firmware repo.

diegoparrilla, 2024-09-13:
The new code base handles better the length of the password, so the issue should be fixed.

closed 2024-09-13 by diegoparrilla