sidecartridge/atarist-sidecart-raspberry-pico #136: Limit on the number of files usable with 'floppies' folder

8bitbill, 2024-10-02:
I dropped all of the "floppyshop" floppies into my sidecart's SD card's "floppies" directory.

But 3,200+ files is too much for the Configurator to currenlty display when using the Emulate a Floppy option.

It did spin on "Loading Floppy Images..." for a while like it was building the list to choose from, but then it ended up only displaying garbage on screen with nothing usable to select.

diegoparrilla, 2024-10-02:
The rp2040 can't support that number of files due to design constraints. 

To workaround the issue:

1. Disable the file counting flag as explained here: https://docs.sidecartridge.com/sidecartridge-multidevice/userguide/#count-the-number-files-in-the-folders-of-the-microsd-card

2. Create multiple folders and split the .st files (ex: /floppies_A /floppies_B ... /floppies_Z). This is how floppy database with thousands of files is remotely stored.

As explained in multiple conversations in the Discord servers, the memory limitations of the device condition the number of files to display. V2 and new microcontroller versions with more memory should relief this limitation.

closed 2024-10-02 by diegoparrilla

aadhoc, 2024-11-14:
After creating the different floppies directories, how do I use them in the UI?
(I was hoping for "Letter navigation" like the "5. Download the Floppy Images database")

Is it a matter of changing the configuration each time like this?

- "C. Configuration Parameters"
-- Change the FLOPPIES_FOLDER to the desired directory (ex: /floppies_M)
-- return to main menu
- "3. Emulate Floppy Disk Drive"
-- Navigate list

> 1. Disable the file counting flag as explained here: https://docs.sidecartridge.com/sidecartridge-multidevice/userguide/#count-the-number-files-in-the-folders-of-the-microsd-card
> 2. Create multiple [...]