sidecartridge/atarist-sidecart-raspberry-pico #141: HD emulation only works for C:

danboid, 2024-10-27:
The HD emulation feature in v0.0.17 of the sidecart firmware only works if you use C: for the drive letter. I have tried D: and E: but neither worked.

I want to be able assign the sidecart to be drive D: because then I might be able to use it to copy files from my SD card onto my UltraSatan which is drive C: That's probably not an option currently because they both want to be drive c:

xdemax, 2024-10-27:
on the latest firmware it is reassigned normally. i have acsi2stm with C to K drive letters. sidecartridge normal work with him. in configurator i assigned gemdrive to L letter.

danboid, 2024-10-27:
You were able to flash the latest sidecart firmware? I tried but it didn't work for me so I opened a ticket about it.

xdemax, 2024-10-27:
yes -- 0.0.18

diegoparrilla, 2024-10-30:
Disk drive assignment applies uniformly to all drive letters. The only reason a drive may not be assigned is if another driver is already using that letter. For instance, UltraSatan and ACSI drivers often reserve multiple letters, even if only one partition is visible.

To avoid conflicts, try using a less common letter, such as 'K'.

Many users report successfully using GEMDRIVE alongside UltraSatan and ACSI2STM with different drivers without any issues. I recommend double-checking your configuration.

closed 2024-10-30 by diegoparrilla