sidecartridge/atarist-sidecart-raspberry-pico #143: STOS can'

NZSPY, 2024-11-10:


diegoparrilla, 2024-11-14:
I can confirm the same behavior with the provided version.

Based on the traces in the GEMDRIVE code, it reads the first ten bytes of the file where the magic word "Lionpoulos" appears, then closes the file and releases the file handle without reporting any errors. My guess is it should continue loading the rest of the .BAS file, but it stops after reading (correctly) the first 10 bytes.

I'll investigate further by enabling additional traces in various GEMDOS calls.

Thank you for the detailed feedback.

NZSPY, 2024-11-14:
Thanks for looking at this, besides playing games on my STE, trying to write some in STOS is my other hobby, and if this could work from the GEM drive. It would mean I don't need to have the Ultra Satan etc.. Can run everything from one Large Fat32 Partition... Rather than been limited to lots of 512kB Fat 16 partitions etc.

I did find a work around.. if I load a .bas file using the US drive with GEMdrive turned off and save it in the .asc format. STOS does load the .ASC file from the GEM drive etc

diegoparrilla, 2024-11-14:
I will continue investigating the issue. It can be an edge case where the file-read GEMDOS call "assumes" that some register is not going to change (or has to change), and the GEMDRIVE implementation modifies the value. 
The bad news is it's hard to debug and fix. The good news is it makes the driver more reliable.

The funny thing is that STOS loads the file header successfully, but it decides that it is not properly formatted.