sidecartridge/atarist-sidecart-raspberry-pico #145: Pico2_w and network features

Fbeen, 2024-12-14:
First of all I would like to express my admiration for the people who made this project. It is really great and I think it will soon be used by many Atari enthusiasts all over the world! Thanks to everyone who contributed to this project.

**I have a few ideas for this project:**
1) perhaps it is possible to actually connect the ST(e) computers to the internet with a tcp/ip stack and a driver as a desktop accessory. Like Sting and NetUSBee, but with the advantage that the hardware is cheaper and more extensive.

2) Judging by the versions of the pico-sdk and pico-extras repositories you started before the pico2-W came out. Support for the pico2 is interesting because it has twice as much ram. Since the firmware reserves 128K for the cartridge, the program ram part becomes three times as large on the pico2. This space is absolutely necessary for the network possibilities.

3) Actually this is the first idea I had. An FTP server running on the pico_W that can also access the files on the ST (including the ST hard disk) via the cartridge port and software on the ST. I have already started this expansion myself, but I am running into memory shortages on the pico_w.

I am curious whether a side branch can be created in which the current firmware can be ported to the pico2_w. I am already working on this locally, but I can certainly use some help, especially from those who know this project.

Cheers!

diegoparrilla, 2024-12-14:
Thank you for your support @Fbeen !

All the topics are in progress. Regarding the Pico 2W and RP235x it will come eventually once the erratum RP2350-E9 is fixed (https://hackaday.com/2024/08/28/hardware-bug-in-raspberry-pis-rp2350-causes-faulty-pull-down-behavior) Since the code changes the direction in the PIO at very high frequency this bug directly impacts all my developments. 

Feel free to work in a branch, of course. All collaborations are welcome. Also, keep in mind that I am developing a version 2 of the firmware that will tackle the problem of custom firmwares. Here are some details I wrote weeks ago: https://sidecartridge.com/blog/2024/11/12/sidecartridge-year-zero/#firmware-roadmap-for-2025

Last, if you feel things do not work well with pico 2W, remember the erratum.  

Also, don't hesitate to hang out on the Discord server: https://sidecartridge.com/contact/

Fbeen, 2024-12-14:
Thanks for the reply [diegoparrilla](https://github.com/diegoparrilla)

Didn't know about the hardware bug in the pico2, what a bummer although there are some workarounds. See you on Discord .