sidecartridge/atarist-sidecart-raspberry-pico #151: Huge crash after updating SidecarT

Niiiko77, 2025-02-20:
Hello everyone, 

I'm posting this because I have a problem that I can't solve. So far I've been using my SidecarT without any problems. I have already carried out updates without any problems and everything was fine. 

This morning I wanted to update the firmware. Everything went well, I followed the procedure but once I plugged it back into the ST it didn't work. 

I had cleaned the cartridge port beforehand. So I reconnected the SidecarT to my PC to carry out an update.

Now it's asking me to format the disk once the SidecarT is plugged in via USB. I did ‘cancel’ to leave it alone, rebooted my PC and reconnected the SidecarT and now it just says ‘USB device unknown’.

I try to find if someone already had that kind of problem...But it seems that I'm the only one? 

I can't do anything, I don't think I've done anything wrong.... I need help!   

Thanks for your replies

diegoparrilla, 2025-02-20:
Check what kind of microSD card you are using. If the microSD is detected when plugging the device via USB as you say, the firmware is running. Use SDHC or SDXC with the newer firmwares: https://discord.com/channels/1160868666162823218/1160869878635429958/1337710589556035628

If necessary, rollback to the previous working version. You can find the previous releases here: https://github.com/sidecartridge/atarist-sidecart-raspberry-pico/releases

Also, keep in mind that the SidecarT won't work in your Atari ST computer when running in mass storage mode: https://youtu.be/h6nc69UldEw?si=x88FMsBhb_NPMT2l You must unplug it first, then power on the Atari ST. As a rule of thumb, the SidecarT won't work with the Atari ST if it does not get the power from the Atari.

Niiiko77, 2025-02-20:
Hi, 

And a great thanks for your help! 

So I try your link to Discord but it doesn't work. I'd love to do a rollback but I have no access to SidecarT who still unrecognized (see photo). I've tried with another SD card, without an S card, on another computer...I still get the same thing and the device isn't recognised.

Yes, I'll do the procedure to exit the mass storage...Without success

[image]

diegoparrilla, 2025-02-20:
The Discord link will work if you have a Discord account: https://discord.com/channels/1160868666162823218/1160869878635429958/1337710589556035628

To upgrade the firmware, you must enter in RPI-RP2 mode: https://docs.sidecartridge.com/sidecartridge-multidevice/how_to/#update-the-firmware

If you're unable to enter RPI-RP2 mode, it likely means your Pico W is severely damaged. This suggests that the microcontroller's internal ROM isn't detecting the hardwired signals required to enter firmware upload mode, which is extremely unusual. I have only seen that damaging physically (cutting paths on the PCB) the pico w.

If you can't enter into RPI-RP2 mode, the most probable reaons are:
1. A windows antivirus playing tricks.
2. A cable failure. This is the number one issue upgrading the firmware by far. 

Try to enter intoi RPI-RP2 mode first, and then copy the firmware dragging and dropping it.

Niiiko77, 2025-02-20:
A few news items. I'm feeling studious because it was indeed the USB cable that was at fault. I've got another one. 

I've rebooted, updated the PICO and SidecarT and all is well again!!!!

Sorry for the loss of time and, above all, a big thank you for your help! 

Best regards
Nicolas

closed 2025-02-20 by Niiiko77

diegoparrilla, 2025-02-20:
No problem, mate! Enjoy!!!