sidecartridge/atarist-sidecart-raspberry-pico #153: Mass storage mode not working

danboid, 2025-04-07:
I am running sidecartridge firmware 1.0.1 with an STe running TOS 1.06 but I have not been able to get the USB mass storage mode working to access files on my sidecartridge SD card.

I have read the mass storage mode instructions here:

https://docs.sidecartridge.com/sidecartridge-multidevice/userguide/#mass-storage-mode

One thing not covered is which USB port to use? Should mass storage work with either of the USB A ports? Both? Maybe mass storage mode only works with the microSD port on the Pico?

I have tried to use all three ports one at a time. In the case of the USB A ports I tried using multiple USB A to USB A cables as well as USB A to USB C cables but nothing worked under both Linux (MATE desktop) or Windows 10. I tried connecting the cable when in the config menu and I also tried having the cable connecte and cold booting into the config menu - nothing.

My micro SD card has a single 2 GB FAT16 partition on it and it works fine with the sidecartridge otherwise eg loading floppy images.

diegoparrilla, 2025-04-07:
USB-A connectors in the board are only power lines, they can't have data lines. Only the micro usb in the pico wh has data lines: https://docs.sidecartridge.com/sidecartridge-multidevice/revisions/#revision-220

There is a video explaining the feature: https://youtu.be/h6nc69UldEw?si=12betqGzk2GeWNEe

Also check that the USB cable has power and data lines. Most of the issues reported were due to faulty cables, or lacking data lines.

diegoparrilla, 2025-04-08:
> Hi Diego
>
> I've got it working now. I had to try five different micro USB cables
> before I found one that supported data.
>
> Its not yor faukt but micro USB is alost extinct now it seems, the RPi
> Pico being one of the very few things to still use it. Although I haveat
> least 5 micro USB cables still, I could not find any other devices in my
> current collection that used micro USB apart from an old Nokia burner I had
> but that only used it for power.
>
> Before we close this ticket, we could do with a few tweaks to the mass
> storage docs. If its on github then I could do a PR?
>
> For
> https://docs.sidecartridge.com/sidecartridge-multidevice/userguide/#mass-storage-mode
>
> As of Beta release v0.0.17, the SidecarTridge multidevice board introduces
> the Mass Storage Mode feature, enabling users to access the microSD card as
> a mass storage device when connected to a computer via USB.
>
> Should read:
>
> As of Beta release v0.0.17, the SidecarTridge multidevice board introduces
> the Mass Storage Mode feature, enabling users to access the microSD card as
> a mass storage device when connected to a computer via the micro USB port.
>
> When properly connected, the small green LED on the Raspberry Pi Pico W/WH
> will stay on steadily.
>
> Whilst the LED on the main sidecart PCB is green, I'd say the LED on the
> pico (which is what is relevant here for the mass storage status) is either
> yellow or amber. Its definitely not green.
>
> We could also add something about a data capable micro USB cable being
> required and that the majority of micro usb cables out there are not?

Thank you For the feedback, Dan!

I will update it to properly. Yes, the micro usb sucks. It’s a cost-driven
decision that makes no sense. USB C has lot more advantages. Actually, my
custom boards in the ROM emulator implement USB C, micro usb is a dead end.

Enviado desde mi móvil. Perdona la brevedad y las faltas de ortografía.

danboid, 2025-04-10:
Unfortunately I only got to use this feature twice before it stopped working. My guess is I might've been a little to heavy handed when inserting or removing a cable from the micro USB socket and now the socket probably needs resoldering.

This is the other problem with the USB micro and USB C to a lesser extent sockets, they can be quite fragile. Not good if you're frequently (un)plugging the cable.

danboid, 2025-04-10:
No wait, it is still working! It wasn't a bad solder joint.

When I am in the sidecart config menu, the sidecart doesn't see to get automatically mounted every time I connect the micro USB cable to my laptop. Sometimes I need to also push the RESET button on the main Sidecart PCB and that causes the SD card to get mounted as a mass storage device.

diegoparrilla, 2025-04-12:
The issue affects the current setup of the user. Not an issue on generic setups.

closed 2025-04-12 by diegoparrilla