sidecartridge/atarist-sidecart-raspberry-pico #154: HD emulator doesn't work after using mass storage mode

danboid, 2025-04-11:
Using sidecart firmware 1.0.1 on an STe with TOS 1.0.6.

Sidecart config menu option 4 "Emulate Hard Disk Drive" doesn't work after using mass storage mode.

* Boot STe into sidecart config menu

* Use mass storage mode (I often I have to push the RESET button to get this to work)

* Select option 4 - emulate Hard Disk Drive

* Push s to start the emulation..... nothing

It should instantly prompt the user to push any key to reset your ST but this takes at least 30seconds to show and when you do push any key after being prompted it resets in TOS but there is no C: drive to be seen.

I should note that after using mass storage mode and unmounting the sidecart undel Linux, Linux sees it as unmounted but the yellow LED on the Pico remains lit and I'd expect that to go off when I unmount it.

If I power cycle the ST and sidecart afterwards, hard drive emulation works fine if I don't use mass storage mode first.

diegoparrilla, 2025-04-12:
The issue relates to the SD card deinitialization process. Firmware version 2 handles this more gracefully using an updated version of the SD card library.

It won’t be addressed in firmware version 1 unless a pull request is submitted. However, it’s unclear whether the current library version used in v1 can support proper deinitialization.

This is a common limitation in many microcontroller libraries — deinitialization is often overlooked or not handled well due to various constraints.

danboid, 2025-04-12:
When can we expect firmware v2?

Another closely related issue that I've not opened an issue for or raised on the forums yet is that I've only been able to get the HD emulation to work as drive c:, not as d: or e: for example.

I have got an UltraSatan using the ICD drivers and when I have that attached it uses c: (and d: and e:) so if I was wanting to use the sidecartridge it would have to work as f: (or d:, if I got rid of the extra 2 ICD partitions).

Is there any hope for me being able to use the sidecart hd emulation at the same time as using my Ultrasatan? Maybe a future revision of the sidecart could removethe need for the ICD (or whatever) drivers and it could mount FAT partitions on ACSI or IDE drives in addition to one on the microSD card?

diegoparrilla, 2025-04-12:
V2 beta will be available in the coming weeks. The alpha version is already downloadable for early testers.

There’s currently no release date for the stable v2.

Gemdrive can operate using any drive letter starting from C. It’s common practice to assign ACSI drives to the lower letters and use Gemdrive with higher ones—this works across all TOS versions.
If Gemdrive fails to allocate a drive letter, it’s likely because another driver is already using it.

Booster app: https://github.com/sidecartridge/rp2-booster-bootloader
Micro firmware template: https://github.com/sidecartridge/md-microfirmware-template
Documentation: https://docs.sidecartridge.com/sidecartridge-multidevice/programming/
https://docs.sidecartridge.com/sidecartridge-multidevice/hardware_interface/

Consider these repos WIP.

danboid, 2025-04-12:
> If Gemdrive fails to allocate a drive letter, it’s likely because another driver is already using it.

I should've said that when I have been testing using alternate (ie not c:) drive letter wth gemdrive, I unplugged my ultrasatan to avoid clashes with gemdrive so its not that causing my problem.

I presume that you have had no problems in assigning the gemdrive other letters? ie d:, e: etc?

I shall try again when the v2beta firmware arrives.

danboid, 2025-04-12:
This caught my eye on the booster github readme:

```
The current version is designed for the RP2040-based Pico W, but future releases will support the RP235x series and move beyond the Atari ST architecture.
```

Are you planning a sidecartridge for another platform?

danboid, 2025-04-13:
> Booster app: https://github.com/sidecartridge/rp2-booster-bootloader Micro firmware template: https://github.com/sidecartridge/md-microfirmware-template Documentation: [...]

I have submitted a PR to add my suggested changes and additions to the mass storage section of the user guide.

danboid, 2025-04-19:
Also, I cannot eject or insert floppy disk images under the sidecart floppy config menu after using mass storage mode under v1.0.1